samphonic/mitsubishi-uart #4: Changes broke compilation

MyGithubUser01, 2024-01-25:
Love the project, I have been using this successfully for a while on two Kaiteki pumps with m5stack c008. However recently it fails to compile/update. To get it to compile I needed to change to "previous release" by setting ref: first-pass in esphome config. 

Is "main" in a working state?

Sammy1Am, 2024-01-25:
Oh no!  I'm so sorry, I honestly had no idea anyone at all was using it or
I'd have been more gentle with the repo changes.

Happy to hear it was working for you though!

Stick with `first-pass` for now, `main` is actually working really well for
me at the moment, but the configuration has changed and there's no
thermostat support yet.  (I'm largely re-writing a lot of it because I'd
been having stability issues with the old version)

I'd hoped to update the documentation either today or tomorrow, and then if
you're not using a thermostat you should be good to switch to main.
Otherwise it'll be a few days before I rebuild the thermostat code.

Sorry again, but hopefully it'll work even better soon.

MyGithubUser01, 2024-01-25:
Vane directional control has been working fine, but I am not really using it.

I am not using a thermostat at the moment, but considering it.

Mostly using automations in Home Assistant to adjust temperature and mode setting.

Let me know if there is anything you'd like me to test/verify. Happy to help.

Sammy1Am, 2024-01-27:
Sorry for the slight delay (whole family got sick so I was unable to code for a bit there).

I've updated the [minimal configuration example](https://github.com/Sammy1Am/mitsubishi-uart/blob/main/esphome-example.yaml), and the [Getting Started](https://github.com/Sammy1Am/mitsubishi-uart/wiki/Getting-Started) page on the wiki to reflect the changes that have now been tagged as `release` instead of `main` to hopefully keep things a little more stable for everyone.  The configuration is just a lot simpler now, but the basic behavior shouldn't be affected.

> Vane directional control has been working fine, but I am not really using it.

This is next on my list (I mostly wasn't using it, but after my testing, the vanes are pointing a direction I don't want, so now I kinda want it 😜 )

> I am not using a thermostat at the moment, but considering it.

This is a big feature for me, so that'll be next.  I want to have some sort of physical control for guests or family members who don't want to bother with Home Assistant.  Plus I've got the Kumo Cloud adapter, and it would be kinda neat to keep that around for now.

> Let me know if there is anything you'd like me to test/verify. Happy to help.

Honestly just knowing that anyone else got this working at all is huge help.  The documentation on how the heat pump works is pretty sparce and anecdotal, so even though it seems to be working for me, I have no idea how it's working for others.

If you run into any issues or weirdness though, definitely open a ticket and let me know.  Likewise if you have any feature requests I'd be happy to take a stab at them as best I can with my setup.

lcoppenr, 2024-01-29:
I'm using it too!   Thank you for your efforts.   I was using a Tasmota solution which was very limited and only had an esp8266 image.  This esphome component is a significant improvement.  Seems to be working well, this spring I will use this for our other 4 units.  The nice thing about the esphome solution is I can add a DS18b20 and sense the temp of the actual air coming from the units which lets me know if there are any issues with the unit.

I too need a physical device to make it easy for family/guests to control heat/cooling.   Our home uses baseboards for heat, so I have a thermostat that I've tricked into heat/cool which then mimics the settings to the AC units for cooling (and secondary heat).  I'm still working out the logic, but the base functionality works.   

For me the top features I'd love to see:
* Vane control (up/down, left/right)
* "Powerful" button like I have on my RF remote, which sets the unit to cool as much as possible for 15min.
* More info about the unit and status.  I don't know what info is available so it is hard for me to guess at what is available via CN105.  Out of curiosity, is model# / serial # available?   I haven't seen a list of available info anywhere, I'll look more at the seed projects.

Again, thanks for the time and effort.   This project has definitely made my life a little easier when it comes to setting up a slighly complex HVAC setup.

Sammy1Am, 2024-01-31:
So happy to hear it's helpful to other people as well!  I just added vane support back to `release`, so unless you need to use an external thermostat, I'd really appreciate if the two of you could let me know if the new codebase is working for you (as mentioned above, there are some config changes, but it's mostly deleting a lot of unneeded config).  Then I'll probably close this issue, and any additional discussion can move somewhere with a more appropriate title.

---

> I can add a DS18b20 and sense the temp of the actual air coming from the units

Hadn't even thought about doing that, that's a neat idea!

> Vane control (up/down, left/right)

Done!  (I don't have a unit with left-right, so I can't test it-- if you get a chance to test I'd love confirmation that's working)

> "Powerful" button like I have on my RF remote, which sets the unit to cool as much as possible for 15min.

I've seen mention of this other places, but no one seems to have found how to turn it on.  My remote doesn't have it, and it doesn't seem like the MHK2 thermostat or Kumo Cloud adapters support this mode, so I don't have the means to sniff the command packets and figure out how to set it.  You could certainly make a Home Assistant automation of sorts to kind of emulate it (setting the temperature as low as it goes and the fan speed all the way up), but I know it's not exactly the same)

> I-See status (this feature is confusing and I'm not sure it is available)

Another one I can't directly test, but it seems like a true/false status might be available (See next item below)

> More info about the unit and status. I don't know what info is available so it is hard for me to guess at what is available via CN105.

So: I've started documenting packets on [this project's wiki](https://github.com/Sammy1Am/mitsubishi-uart/wiki/Decoding-Packets), but haven't even gotten around to documenting all the basic packets that we *do* support (actually getting the code working has been a higher priority).  There's a bunch of links at the top of that page that point to other places where people are trying to discover packet behavior.

Previously, this project had support for "Loop Status", "Stage", and "Compressor Frequency", but I haven't re-added them yet because it's not super clear what the values really mean.  E.g. I only ever saw stages 0, 3, and 4, and status 0, and 4.  I'm not sure compressor frequency ever reported a value on my system, and I don't know what the value means.

My long-term goal is to eventually add support for as many of these as I can pin-down meaning-wise; but if you:
<br>**a)** have one that you particularly would like added I can stick it in there for you (with just raw values), or 
<br>**b)** are willing to try figuring it out on your own, I can help with logging/dumping packet data so you can look for patterns.

And again, thanks for just stopping by to say you're using it.  I've been really happy with Mitsubishi's hardware, but the controls are a bit bonkers and this seemed like a good way to get a better handle on the system.

lcoppenr, 2024-01-31:
I installed the new release tonight and everything appears to be working correctly.   Horizontal and Vertical vane control was quite responsive.    Success!

Honestly I think most of the other features are fringe features and are just "nice to have"s.  I'm sure for diagnostics they would be helpful, but I doubt most of the audience for this type of thing is actually performing diagnostics.  The temp sensor at the output is probably the most relevant information for a homeowner to pass on to a technician (plus ambient temp).

Focusing on repeating/extending the cn105 to the Kumo or MHK2/MHK1 may be the best use of your time.    I can help in any way I can.   Just let me know how I can help.  Is there a way to do a log/dump through the ESP32 or do I have to hook up something more powerful?

BTW: The unit I am testing with is an MSZ-FE12NA

Sammy1Am, 2024-02-01:
Excellent, happy to hear it's working.

> Is there a way to do a log/dump through the ESP32 or do I have to hook up something more powerful?

Setting the ESPHome `logger:` configuration to `VERBOSE` will print out all the packets being received from the heatpump to the ESPHome logs.  I just committed a change to the `main` branch here that colorizes them too for easier reading.

At the moment, you'll only really be getting the 3 well-known status packets (but you could still look at those to e.g. see if there's any iSee information in there).  My plan is to get the thermostat support added back in (so it can capture/log packets sent from the MHK2), and then if things are still stable, hook one up between my MHK2 and Kumo to see what they're talking about.  Then depending on how that goes, I might put it between the Kumo and the airhandler, or leave it as it (still trying to decide if it makes more sense for the microcontroller or the Kumo to be directly connected).

I've only tested on two MSZ-GL06NA, but once I feel confident it won't break and inconvenience the family, I'll be trying it on the SVZ-KP30NA that does the rest of the house.

Probably nothing in particular for you to test at the moment; once thermostat support is there, you could try searching more for stuff like iSee or "Powerful" if those are available on an MHK or Kumo.

Sammy1Am, 2024-02-07:
`v1.2.0` adds thermostat support for the MHK2 (and in theory Kumo, but I haven't tested yet).  This brings the new codebase up to parity with where it was before I broke things.

It sounds like things are working okay so I'm going to close this issue for now.  If a new issue arises with compilation, or there's a feature request or something, please feel free to open a specific ticket for it.

closed 2024-02-07 by Sammy1Am