samphonic/mitsubishi-uart #5: Cannot compile

tjwallace, 2024-02-07:
Config (`hp_test.yaml`):
```yaml
substitutions:
  devicename: esp-hp-test
  upper_devicename: ESP HP Test

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini

# wifi, etc setup
<<: !include common.yaml

external_components:
  - source: github://Sammy1Am/mitsubishi-uart@release

mitsubishi_uart:
  heatpump_uart: uart_hp

uart:
  - id: uart_hp
    baud_rate: 2400  # 9600 for some systems
    parity: EVEN
    rx_pin:
      number: GPIO3
    tx_pin:
      number: GPIO1

```

Log:
```
$ esphome run hp_test.yaml

INFO ESPHome 2023.12.9
INFO Reading configuration hp_test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp-hp-test (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
INFO Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@2.0.0 has been installed!
INFO ESPAsyncTCP-esphome@2.0.0 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@3.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing ottowinter/ESPAsyncTCP-esphome
INFO Installing ottowinter/ESPAsyncTCP-esphome
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP-esphome@1.2.3 has been installed!
INFO ESPAsyncTCP-esphome@1.2.3 has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
Compiling .pioenvs/esp-hp-test/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/esp-hp-test/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esp-hp-test/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/esp-hp-test/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/esp-hp-test/src/esphome/components/api/api_server.cpp.o
[... 101 lines cut ...]
  262 |   ESP_LOGD(TAG, packet.to_string().c_str());
      |   ^~~~~~~~
src/esphome/components/mitsubishi_uart/mitsubishi_uart-packetprocessing.cpp:262:41: error: array must be initialized with a brace-enclosed initializer
  262 |   ESP_LOGD(TAG, packet.to_string().c_str());
/Users/jeff/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
src/esphome/core/log.h:70:36: note: in expansion of macro 'F'
   70 | #define ESPHOME_LOG_FORMAT(format) F(format)
      |                                    ^
src/esphome/core/log.h:95:59: note: in expansion of macro 'ESPHOME_LOG_FORMAT'
   95 |   esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
      |                                                           ^~~~~~~~~~~~~~~~~~
src/esphome/core/log.h:152:28: note: in expansion of macro 'esph_log_d'
  152 | #define ESP_LOGD(tag, ...) esph_log_d(tag, __VA_ARGS__)
      |                            ^~~~~~~~~~
src/esphome/components/mitsubishi_uart/mitsubishi_uart-packetprocessing.cpp:262:3: note: in expansion of macro 'ESP_LOGD'
  262 |   ESP_LOGD(TAG, packet.to_string().c_str());
      |   ^~~~~~~~
Compiling .pioenvs/esp-hp-test/src/esphome/components/network/util.cpp.o
*** [.pioenvs/esp-hp-test/src/esphome/components/mitsubishi_uart/mitsubishi_uart-packetprocessing.cpp.o] Error 1
===================================================================================================== [FAILED] Took 5.83 seconds =====================================================================================================
```

The only tag I could get to compile was [`first-pass`](https://github.com/Sammy1Am/mitsubishi-uart/releases/tag/first-pass). Let me know if I can help debug this. Thanks!

Sammy1Am, 2024-02-07:
I get the same errors trying to target that board on my end, so I'll work on a fix as soon as I can.

(If you've got an esp32 in the meantime you might have better luck)

closed 2024-02-07 by tjwallace

Sammy1Am, 2024-02-07:
Great, glad it works!  Let me know of it goes when you get them connected.

tjwallace, 2024-02-11:
I tested the device and it works well! Instant feedback and updates. Great project, thanks again!

Sammy1Am, 2024-02-12:
Awesome, thanks for letting me know!