samphonic/mitsubishi-uart #6: Vane in unknown horizontal position 7

tjwallace, 2024-02-14:
I have 3 MSZ-GL06NA wall units which do not have controllable horizontal vanes.

I have disabled the select option like this:

```yaml
mitsubishi_uart:
  heatpump_uart: uart_hp
  selects:
    horizontal_vane_position_select:
      id: horizontal_vane_position_select
      disabled_by_default: true
```

But in the logs I see this:

```
[14:25:49][W][mitsubishi_uart:167]: Vane in unknown horizontal position 7
```

I think I can just ignore the log message but could an option be added [here](https://github.com/Sammy1Am/mitsubishi-uart/blob/main/components/mitsubishi_uart/muart_packet.h#L251-L259) for `0x07`?

Sammy1Am, 2024-02-16:
That's fascinating.  First: yes, you can safely ignore those warnings; they're mostly just to say "careful, this code doesn't know what this number means".

Do all three units show the same `7`/`0` behavior?  I also have two MSZ-GL06NA units, but the one I just checked is showing the horizontal position as `3` when on, and also `3` when off.  **However**, I'm almost certain at some point I "changed" the position to 3.  So maybe `7` is some sort of factory default that means "not supported"?

My only hesitation with adding them there is I'm not sure what happens if we send a command to the unit to set the horizontal vanes to `7` or to `0`.  Presumably for your units nothing bad would happen since they're already showing that.  I can try mine later and see what it does (probably nothing).  Like, chances of it breaking something seem really slim, but damage caused by the warning message in the logs is also pretty slim.

I'll leave this open for now and maybe do some experimenting here / researching to see if anyone with actual horizontal vanes has anything to add.  As a compromise I could maybe suppress the warning for those two cases as sort of like, "semi-known, but we don't support switching to them"?

tjwallace, 2024-02-17:
I will investigate if all 3 units report the same values. However if they are fine to ignore it probably isn't worth making the code more complicated by adding the values to ignore - what do you think? I'm also happy to leave this open to track our research.

Sammy1Am, 2024-02-17:
Agree on not adding code unless research turns up what the values actually mean.  Let's leave ticket open until then (like, presumably they mean *something* since it seems like your units came with those values without any meddling from this code).

Olgmo, 2024-04-24:
Hi, I am new here and testing the code on my unit.
I have a MSZ-LN35 and run the code on a M5Stack atom lite in latest ESPHome and HomeAssistant.
I have these errors:
"[W]	[mitsubishi_uart:183] Vane in unknown horizontal position 81" etc
Tested the horizontal position from from left to right and I got these codes.

81, 82, 83, 84, 85, 88, 8c
where 83 is centre and 88 is split and 8c is auto
I have aslo seen position 80, but not sure what that is.

KazWolfe, 2024-04-24:
@Olgmo Very very interesting - it looks like 8x might be a flag of some sort, but I have no idea what it means just yet. More research time!

If you set the horizontal vanes via UI as-is, does it work? Here's a little test case:

1. Set the horizontal vane to fully left via remote (should be 81), verify the packet arrives.
2. Set the horizontal vane to fully right via Home Assistant, check what packet is returned. Does the vane actually move, and does the system now report the vane in 05 or 85?
3. Set the horizontal vane to center via remote. Does it report 03 or 83?

Thanks!

Olgmo, 2024-04-24:
@KazWolfe 
reply to 1: (Setting vane from remote controll)
`19:27:16	[W]	[mitsubishi_uart:183]	Vane in unknown horizontal position 83
19:27:35	[D]	[homeassistant.sensor:024]	'sensor.stue_atc_a293_temperature': Got state 21.77
19:27:35	[D]	[sensor:094]	'stue_temp': Sending state 22.77000  with 1 decimals of accuracy
19:27:35	[I]	[mitsubishi_uart:276] Received temperature from stue_temp of 22.770000. (Current source: stue_temp)
19:27:35	[I]	[mitsubishi_uart:020] Generic unhandled packet type 61 received.
19:27:35	[D]	[mitsubishi_uart:021]  [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m
19:27:46	[W]	[mitsubishi_uart:183]	Vane in unknown horizontal position 81
`

(And here is setting horizintal vane from web interface, because I dont see any vane controll i HomeAss)
`19:40:01	[W]	[web_server_idf:070] Only application/x-www-form-urlencoded supported for POST request
19:40:01	[D]	[select:062] 'Horizontal Vane Position' - Setting
19:40:01	[D]	[select:115] 'Horizontal Vane Position' - Set selected option to: <<
19:40:01	[D]	[select:015] 'Horizontal Vane Position': Sending state << (index 1)
19:40:01	[I]	[mitsubishi_uart:020] Generic unhandled packet type 61 received.
19:40:01	[D]	[mitsubishi_uart:021] [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m
19:40:16	[W]	[mitsubishi_uart:183] Vane in unknown horizontal position 81
`

On the side question: So how to get the vane (vertical and horizontal) control into HomeAssistant?

KazWolfe, 2024-04-24:
@Olgmo A config option should be published by default, named `select.horizontal_vane_position_select` and `select.vane_position_select` for that specific device. It usually likes hiding under the device view, and isn't in the climate entity. 

Regarding your packet report: it looks like you set both to left. Can you repeat with the ESP sending a command to the *right* (so `>>`)? I'm trying to see if your heat pump will report 05 or 85. If the former, then we're going to have to figure out what that flag is. If you're reporting 85, then we can just be a bit lazy and mask out the unnecessary bits until we've figured out what that flag is and how it works. This is also why I'd prefer left (IR), right (ESP), center (IR) - it would let me check the state of that bit across requests a bit better.

I'm hoping to get a system with horizontal vanes installed soon as well, which will let me test this sorta stuff in more depth. Interestingly enough, I have failed to find any ability to control the wide vanes via Kumo Cloud so we're flying completely blind here.

Olgmo, 2024-04-24:
Setting Vane to the right from ESP
`
20:10:02 [W] [web_server_idf:070] Only application/x-www-form-urlencoded supported for POST request
20:10:02 [D] [select:062] 'Horizontal Vane Position' - Setting
20:10:02 [D] [select:115] 'Horizontal Vane Position' - Set selected option to: >>
20:10:02 [D] [select:015] 'Horizontal Vane Position': Sending state >> (index 5)
20:10:02 [I] [mitsubishi_uart:020] Generic unhandled packet type 61 received.
20:10:02 [D] [mitsubishi_uart:021]  [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m
20:10:05 [D] [esp32.preferences:114] Saving 1 preferences to flash...
20:10:05 [D] [esp32.preferences:143] Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
20:10:11 [D] [homeassistant.sensor:024] 'sensor.stue_atc_a293_temperature': Got state 21.86
20:10:11 [D] [sensor:094] 'stue_temp': Sending state 22.86000  with 1 decimals of accuracy
20:10:11 [I] [mitsubishi_uart:276] Received temperature from stue_temp of 22.860001. (Current source: stue_temp)
20:10:11 [I] [mitsubishi_uart:020] Generic unhandled packet type 61 received.
20:10:11 [D] [mitsubishi_uart:021]  [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m
20:10:16 [D] [homeassistant.sensor:024] 'sensor.stue_atc_a293_temperature': Got state 21.88
20:10:16 [D] [sensor:094] 'stue_temp': Sending state 22.88000  with 1 decimals of accuracy
20:10:16 [I] [mitsubishi_uart:276] Received temperature from stue_temp of 22.879999. (Current source: stue_temp)
20:10:16 [W] [mitsubishi_uart:183] Vane in unknown horizontal position 85
20:10:17 [I] [mitsubishi_uart:020] Generic unhandled packet type 61 received.
20:10:17 [D] [mitsubishi_uart:021] [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m
`

And setting LEFT from Home Assistant (Sorry I was not aware of those to select entities)
`
20:19:31 [D] [select:062] 'Horizontal Vane Position' - Setting
20:19:31 [D] [select:115] 'Horizontal Vane Position' - Set selected option to: <<
20:19:31 [D] [select:015] 'Horizontal Vane Position': Sending state << (index 1)
20:19:32 [I] [mitsubishi_uart:020] Generic unhandled packet type 61 received.
20:19:32 [D] [mitsubishi_uart:021]  [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m
20:19:37 [D] [homeassistant.sensor:024] 'sensor.stue_atc_a293_temperature': Got state 22.25
20:19:37 [D] [sensor:094] 'stue_temp': Sending state 23.25000  with 1 decimals of accuracy
20:19:37 [I] [mitsubishi_uart:276] Received temperature from stue_temp of 23.250000. (Current source: stue_temp)
20:19:37 [I] [mitsubishi_uart:020] Generic unhandled packet type 61 received.
20:19:37 [D] [mitsubishi_uart:021]  [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m
20:19:46 [W] [mitsubishi_uart:183] Vane in unknown horizontal position 81
`
Please let me know if I can provide more data.

KazWolfe, 2024-04-24:
Interesting, so it looks like your unit does just prepend the 0x80 for whatever reason. I'm assuming that when you set your vane to the right, it actually did move to the right?

Assuming this works, we should just be able to get the actual vane status very simply, though I suspect that bit still has some kind of indication that we'll need to reverse engineer entirely. SWICago seems to handle it, but they just keep it around as a static set...

Olgmo, 2024-04-24:
yeah, the vane works correctly both from HA and ESP web.

KazWolfe, 2024-04-24:
@Olgmo I've opened PR #38 to start to track what the MSB field is. If you're willing to try experimental code, you can use this branch:

```yml
external_components:
  - source:
      type: git
      url: https://github.com/KazWolfe/mitsubishi-uart
      ref: hvane-msb
```

I'll add more documentation fields in the future once I get my own unit to test, but I'm curious to see if this'll clean up the errors and restore functionality for you.

Olgmo, 2024-04-25:
Tested with your TEST source code:
_____________________________________
Setting VANE from HomeAssistant

09:43:41 [W] [web_server_idf:070] Only application/x-www-form-urlencoded supported for POST request
09:43:41 [D] [select:062] 'Horizontal Vane Position' - Setting
09:43:41 [D] [select:115] 'Horizontal Vane Position' - Set selected option to: <<
09:43:41 [D] [select:015] 'Horizontal Vane Position': Sending state << (index 1)
09:43:41 [I] [mitsubishi_uart:020] Generic unhandled packet type 61 received.
09:43:41 [D] [mitsubishi_uart:021]  [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m
__________________________________
Changing VANE from remote

09:58:31 [I] [mitsubishi_uart:020] Generic unhandled packet type 61 received.
09:58:31 [D] [mitsubishi_uart:021]  [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m
09:58:53 [D] [climate:396] 'kaiteki' - Sending state:
09:58:53 [D] [climate:399]   Mode: HEAT
09:58:53 [D] [climate:401]   Action: IDLE
09:58:53 [D] [climate:404]   Fan Mode: LOW
09:58:53 [D] [climate:419]   Current Temperature: 23.00°C
09:58:53 [D] [climate:425]   Target Temperature: 22.00°C
09:58:53 [D] [select:015] 'Vane Position': Sending state 5 (index 5)
09:58:53 [D] [select:015] 'Horizontal Vane Position': Sending state < (index 2)
__________________________________
Changing VANE from ESP web

10:00:46 [W] [web_server_idf:070] Only application/x-www-form-urlencoded supported for POST request
10:00:46 [D] [select:062] 'Horizontal Vane Position' - Setting
10:00:46 [D] [select:115] 'Horizontal Vane Position' - Set selected option to: >
10:00:46 [D] [select:015] 'Horizontal Vane Position': Sending state > (index 4)
10:00:46 [I] [mitsubishi_uart:020] Generic unhandled packet type 61 received.
10:00:46 [D] [mitsubishi_uart:021]  [0;36m[FC. [1;36m61. [0;36m01.30.10] [0;37m00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00  [0;32m5E [0m

KazWolfe, 2024-04-25:
Alright, assuming the vane actually moves as well, I think then we're good on this one. Going to go ahead and guess that your errors are also gone, @Olgmo?

If you're willing to test some more, can you set your hvane to something like far left and then use HA/ESP to set to "Auto"? I'd be curious if (a) the setting actually persists after ~30 seconds, and (b) if the vane were to actually move in response to that.

Olgmo, 2024-04-25:
@KazWolfe So far I can see, I can operate from both HA and remote controll and the setting presists over time.
No errors..

One other thing. but it not related to this topic is the fan speed.  
The "actual fan speed" on the left (Low) is different from the log on the right (High) See attached image . And in HA it is Medium. 
[image]

KazWolfe, 2024-04-25:
@Olgmo The fan speed issue is being tracked in #27. The names for "actual fan speed" were extracted from Kumo, which don't actually match the "set" fan speed names for... reasons. It's a complicated mess and we haven't figured out a good solution to it quite yet since the mapping doesn't always appear to be 1:1. More info is over in that thread if you're curious.

Sammy1Am, 2024-05-09:
Just merged #38 which should at least suppress the warnings for values like `8x`.  Still not sure what's going on with `0x00` and `0x07`