samphonic/mitsubishi-uart #7: Alternate temperature source seems to have no effect

tjwallace, 2024-02-21:
I have configured `mitsubishi_uart` to use an alternative temperature source but it doesn't seem to have any effect. The internal temperature source is still used. Any idea how to debug?

```yaml
sensor:
  - id: room_temp
    platform: homeassistant
    entity_id: sensor.snzb_02d_2_temperature
    name: Kitchen Temperature

mitsubishi_uart:
  heatpump_uart: uart_hp
  temperature_sources:
    - room_temp
  supported_fan_modes:
    - AUTO
    - LOW
    - MEDIUM
    - HIGH
  custom_fan_modes: []
  selects:
    vane_position_select:
      id: vane_position_select
      disabled_by_default: true
    horizontal_vane_position_select:
      id: horizontal_vane_position_select
      disabled_by_default: true
```

[image]
[image]

tjwallace, 2024-02-21:
Hmm this might just be a timing issue and the climate value doesn't change until the sensor publishes a value

Sammy1Am, 2024-02-21:
For debugging, if the esphome logging is set to at least `INFO`, you should see log messages like `Received temperature from Kitchen Temperature of 22.4. (Current source: Kitchen Temperature)` whenever data is published by the sensor.

But as I think you've discovered: the temperature is only updated when the sensor publishes a value.  This is primarily as a fail-safe so that if the sensor stops working (battery dies, etc.) at least the heat pump can fall back to its internal temperature (rather than running forever or never running).

For sensors defined in esphome, you can set the `force_update` configuration variable to `true` to force Home Assistant "to create a state changed event when the sensor updates even if the value stayed the same."  This state changed event seems to be what causes the `platform: homeassistant` sensors to trigger the `on_state_callback` method and notify mUART of the sensor value.

If this sensor is defined otherwise in Home Assistant, I don't know for sure if there's a similar forcing option to ensure a datapoint is generated every less-than-7-minutes even if the temperature doesn't change.

---

This is mostly unhelpful, but: If the sensor takes too long to report a value, the code will switch back to the internal sensor, but when the sensor eventually *does* report a value, the code will switch back to it... for 7 minutes before switching back to the internal one.  So a brief outage should be automatically recovered from, but that doesn't really work for normal use.

tjwallace, 2024-02-21:
Thanks for the detailed explanation. That all checks out. The sensor is a Sonoff SNZB-02D and unfortunately it doesn't seem to update that often. I have been testing the SNZB-02P and it seems to update much more often so I may get a few more of those.

closed 2024-02-21 by tjwallace

Sammy1Am, 2024-02-21:
This just occurred to me as a workaround: you could define a [template sensor](https://esphome.io/components/sensor/template.html) in the esphome configuration for the thermostat and have the lambda get and return the value of the Sonoff (but with `force_update`) on a regular interval.  This introduces risk if the sensor stops working, but would let you use your existing hardware at least.