samphonic/mitsubishi-uart #14: Another esphome implementation

kkopachev, 2024-03-29:
Just wanted to let you know there was another attempt at native esphome component with hardware serial on this: https://github.com/esphome/esphome/pull/5265

Sammy1Am, 2024-03-29:
Oh!  Thank you for sharing that.  Is it bad form to just go over there and drop a link to this repo?  😉 

I see you did some testing with that attempt, have you tried this project yet?  And if so, is there anything the other one had that we're missing here?

I would love to eventually have this merged into ESPHome so that it's as easy as possible for people to use, but also with the rapid development lately it's sorta nice to be able to iterate quickly without annoying anyone over there.

Still... shouldn't be too hard to just pull this repo into an EPSHome fork for PRs (might even be automatable in GitHub).

KazWolfe, 2024-03-30:
I definitely want to push for an official submission at some point. There are currently more than a handful of projects (most of which linked in the wiki) that seem to have duplicated a lot of effort - it'd be nice for us to back something.

My vote is this project since I kinda like the slickness of the MHK2 implementation, but I definitely do want to backport a lot of our findings to SwiCago too.

sle118, 2024-04-01:
I have been looking for the possibility of doing this ever since I figured Honeywell had a shitty support for setting the fan speed (i.e. didn't bother) with their red link system. How disconnected can a company be to let down it's trail beating customers with something as basic as allowing fan speed to be set. Honestly, this is the single most wanted feature in an imperfect mechanical system (as it is often the case with residential HVAC) that someone need monitoring and adjustment when some rooms are too far from the set point.

I was going to assemble something together, but I think you are giving a better starting point.