samphonic/mitsubishi-uart #31: [MSZ-GE35VA] Setpoint temperature can be read back from the AC but cannot be set

Ricc68, 2024-04-07:
If I set the temperature from the IR remote, the component reads the set temperature correctly from the AC.
If I set the temperature from the component, the AC always receives a setpoint of 31 °C, in fact in this case the readback from the AC is always 31 °C:

[image]

Let me know you want the serial log.

Sorry ... two bugs in less than 10 minutes ... it's an habit that I can't help ... looks like I already care for this component to improve 😊

Sammy1Am, 2024-04-07:
We're just grateful to have someone with different hardware to test on :)

I suspect this is related (there are two different ways for handling temperature, and we've only been testing one of them so far).  I'll do what I can to take a look at this by tonight. 

In the meantime, keep the issues coming if you find anything else :)

KazWolfe, 2024-04-07:
Hi!

If possible, I'd like to get some packets from you. Specifically:

* The `0x7B` Extended Connect Response from your heat pump.
* A `0x62` `0x02` Get Current Settings from your heat pump.
* A `0x62` `0x03` Get Room Temperature from your heat pump.

Right now, I'm going to agree with Sammy that your unit doesn't have support for what Mitsubishi calls the "A" scale of temperature units. While this was known, I'm still interested in seeing how this works and getting some additional information about how certain packets behave under these conditions.

The easiest way to get the 0x7B packet is to (unfortunately) reboot your ESP and wait for the connection handshake again. You're also more than welcome to join us in the Discord for further assistance/faster troubleshooting.

Ricc68, 2024-04-07:
No problem for the reboot, today I did it tens of times, here it goes.

0x7B:

[18:52:45][V][muart_bridge:036]: Sending to heatpump [FC.5B.01.30.01]C9 AA
[18:52:46][V][muart_bridge:013]: Parsing 7b heatpump packet
[18:52:46][V][mitsubishi_uart:046]: Processing Extended Connect Response: [FC.7B.01.30.10]C9.03.00.09.04.14.07.75.00.00.00.00.00.00.00.00 DB
 HeatDisabled:No SupportsVane:Yes SupportsVaneSwing:Yes DryDisabled:No FanDisabled:No ExtTempRange:No AutoFanDisabled:No InstallerSettings:No TestMode:No DryTemp:No StatusDisplay:No
 CoolDrySetpoint:-64.000000/-64.000000 HeatSetpoint:-64.000000/-64.000000 AutoSetpoint:-64.000000/-64.000000 FanSpeeds:4
[18:52:46][I][mitsubishi_uart:052]: Received heat pump identification packet.

0x62 0x02:

[18:53:15][V][muart_bridge:036]: Sending to heatpump [FC.42.01.30.01]02 8A
[18:53:16][V][muart_bridge:013]: Parsing 62 heatpump packet
[18:53:16][V][mitsubishi_uart:064]: Processing Settings Response: [FC.62.01.30.10]02.00.00.00.03.09.05.00.00.00.03.00.00.00.00.00 47
 Fan:05 Mode:03 Power:Off TargetTemp:22.000000 Vane:00 HVane:03
 PowerLock:No ModeLock:No TempLock:No

0x62 0x03:

[18:53:13][V][muart_bridge:036]: Sending to heatpump [FC.42.01.30.01]03 89
[18:53:13][V][muart_bridge:013]: Parsing 62 heatpump packet
[18:53:13][V][mitsubishi_uart:186]: Processing Current Temp Response: [FC.62.01.30.10]03.00.00.0D.00.00.00.00.00.00.00.00.00.00.00.00 4D
 Temp:14.500000

KazWolfe, 2024-04-07:
Ooo, very interesting! This confirms a few things for us, thank you! I've gone ahead and opened PR #33 that should fix this issue (hopefully). In the meantime, I'll add your information to our wiki.

Ricc68, 2024-04-07:
@KazWolfe please see last comments in issue #30 (now closed) in which I mentioned also a lookup table, but better a calculation, on the unit's get/set setpoints and the theory that on my unit the setpoints can only be integer, no fractionals, that I think is useful in resolving also this issue.

KazWolfe, 2024-04-07:
Re legacy setpoint temperatures: we were sending 0x00 over the wire for the legacy temperature setpoint (well, really, we were only setting the A-scale setpoint), so it'd make sense for that to set it to 31.

The Mitsubishi docs again report:
```js
set_temp: function (e) {
  var t = 15 & e,
  n = 240 & e,
  r = 31 - t + (n ? 0.5 : 0);
  return {
    celsius: r
  }
},
```

That is to say, the "whole" numbers are stored in bits 0-3, and the number is assumed to be fractional if any of bits 4-7 are set. 

I'd hazard a guess now (or, at least, I strongly hope that this is the case) that your unit will just ignore bits 4-7 entirely.

Ricc68, 2024-04-07:
Ok, I see your point and it makes sense to me. If you commit to main, tomorrow I will test the set/get setpoints as well.

KazWolfe, 2024-04-07:
It's [already in main](https://github.com/Sammy1Am/mitsubishi-uart/commit/e4515a7dff6f6aabcd797732b70faddb1353ac9b#diff-5ae9b22ff2d7191fc0f78f6518bc80a45302891b0f0942c7d1bed2a0891fab9fR128), so whenever you have time we'll be fine.

Ricc68, 2024-04-08:
I have just tried to compile the main branch but it doesn't compile spitting this error:

In file included from src/esphome/components/mitsubishi_uart/mitsubishi_uart.h:11,
                 from src/esphome/components/mitsubishi_uart/muart_utils.h:3,
                 from src/esphome/components/mitsubishi_uart/muart_packet.h:7,
                 from src/esphome/components/mitsubishi_uart/muart_packet-derived.cpp:1:
src/esphome/components/mitsubishi_uart/muart_bridge.h:20:54: error: 'PacketProcessor' has not been declared
   20 |     MUARTBridge(uart::UARTComponent *uart_component, PacketProcessor *packet_processor);
      |                                                      ^~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_uart/muart_bridge.h:23:27: error: 'Packet' does not name a type; did you mean 'RawPacket'?

I tried to compile also the release 1.3.0 but it's giving the same PacketProcessor not declared error.

KazWolfe, 2024-04-08:
@Ricc68 should be fixed as of #36. Unsure how that got in there, my local environment didn't complain at all during testing...

Ricc68, 2024-04-08:
👍 as soon as the code is pulled in main I will compile and test it.

Sammy1Am, 2024-04-08:
Merged.

Yeah, I also didn't have any trouble compiling (maybe my environment is being extra forgiving or something) to test or I wouldn't have released it.  Thanks for reporting it.

Ricc68, 2024-04-08:
@KazWolfe I have tested the setpoints and it is not working, still jumping to 31 °C when I try to set a temperature.
For example I tried to set 20 °C but in the protocol I see:

[20:20:09][V][muart_bridge:036]: Sending to heatpump [FC.41.01.30.10]01.04.00.00.00.00.0B.00.00.00.00.00.00.00.A8.00 C6

Byte 5 is 00 so seems that it's not encoding the 20 °C to what it should be.

[image]

KazWolfe, 2024-04-08:
God, I'm sloppy apparently. Fix in #37.

Ricc68, 2024-04-08:
That's something I was suspecting also earlier but this time I looked closer at the protocol chat.
Soon as the fix is merged I can test it 👍

Sammy1Am, 2024-04-08:
Merged.  Sorry I'm not catching any of these, but I don't have a unit that's paying attention to those temperature values.

Ricc68, 2024-04-08:
Basically this last one was a byte position error in sending the set legacy temperature parameter so instead of sending the setpoint I chosen it was always sending 0x00 on the wire which the unit translates to 31 °C regardless what I set in the UI.
I'll test two things: 1 setting an integer setpoint which I think now should be per protocol spec so I expect it to work and 2 setting a parameter with decimals like 20.5 °C which we don't know how the unit will react.
I will test this one tomorrow.

KazWolfe, 2024-04-09:
Nice, I'll leave to Sam to close. I'm glad your unit doesn't complain about the setting at the very least, would be curious to see if it actually uses that fractional setpoint properly as well, but that's a bit of a stricter test (and may be trickier in general based on how accurate the heat pump is in general).

Ricc68, 2024-04-09:
What test you were thinking?

If the test is to set 0.5 °C difference from ambient temperature and see if the unit goes there, I'm not sure the home environment is sufficiently stable to appreciate such small change.
If the test is on the protocol then surely I can sniff and analyze the protocol.

KazWolfe, 2024-04-09:
Yeah, you’ve basically identified it. The best test would be to see if the setpoint behaves, but that’s not that simple in general.

The only other protocol test I can think of is checking if the remote still works and will drop it back to a non-fractional value.

Ricc68, 2024-04-09:
I tried to see how the unit behaves with a 0.5 °C setpoint change but it did apparently nothing as the led of "out of setpoint" did not lit.
About the remote it still works and when setting the setpoint from remote it drops back to the non fractional setpoint transmitted by the remote, as expected.