samphonic/mitsubishi-uart #32: [MSZ-GE35VA] Timeout waiting for response to 42 packet.

Ricc68, 2024-04-07:
I keep getting a timeout on one of the 42 commands, I think it is sub-command 09 (sorry if I misunderstood the protocol by just looking at the logs, I have not yet looked at the protocol specs):

[18:49:11][V][muart_bridge:036]: Sending to heatpump [FC.42.01.30.01]09 83
[18:49:14][W][muart_bridge:046]: Timeout waiting for response to 42 packet.

I get an answer for all other sub-commands in the cycle:

[18:56:31][W][muart_bridge:046]: Timeout waiting for response to 42 packet.
[18:56:31][V][muart_bridge:036]: Sending to heatpump [FC.42.01.30.01]03 89
[18:56:31][V][muart_bridge:013]: Parsing 62 heatpump packet
[18:56:31][V][mitsubishi_uart:186]: Processing Current Temp Response: [FC.62.01.30.10]03.00.00.0D.00.00.00.00.00.00.00.00.00.00.00.00 4D
 Temp:14.500000
[18:56:31][W][component:214]: Component mitsubishi_uart took a long time for an operation (0.06 s).
[18:56:31][W][component:215]: Components should block for at most 20-30ms.
[18:56:31][V][muart_bridge:036]: Sending to heatpump [FC.42.01.30.01]04 88
[18:56:31][V][muart_bridge:013]: Parsing 62 heatpump packet
[18:56:31][V][mitsubishi_uart:292]: Processing Error State Response: [FC.62.01.30.10]04.00.00.00.80.00.00.00.00.00.00.00.00.00.00.00 D9
 Error State: No ErrorCode: 8000 ShortCode: A0(00)
[18:56:31][W][component:214]: Component mitsubishi_uart took a long time for an operation (0.06 s).
[18:56:31][W][component:215]: Components should block for at most 20-30ms.
[18:56:33][V][muart_bridge:036]: Sending to heatpump [FC.42.01.30.01]02 8A
[18:56:34][V][muart_bridge:013]: Parsing 62 heatpump packet
[18:56:34][V][mitsubishi_uart:064]: Processing Settings Response: [FC.62.01.30.10]02.00.00.00.03.09.05.00.00.00.03.00.00.00.00.00 47
 Fan:05 Mode:03 Power:Off TargetTemp:22.000000 Vane:00 HVane:03
 PowerLock:No ModeLock:No TempLock:No
[18:56:34][W][component:214]: Component mitsubishi_uart took a long time for an operation (0.06 s).
[18:56:34][W][component:215]: Components should block for at most 20-30ms.
[18:56:34][V][muart_bridge:036]: Sending to heatpump [FC.42.01.30.01]06 86
[18:56:34][V][muart_bridge:013]: Parsing 62 heatpump packet
[18:56:34][V][mitsubishi_uart:196]: Processing Status Response: [FC.62.01.30.10]06.00.00.06.00.00.00.00.00.00.00.00.00.00.00.00 51
 CompressorFrequency: 6 Operating: No
[18:56:34][W][component:214]: Component mitsubishi_uart took a long time for an operation (0.05 s).
[18:56:34][W][component:215]: Components should block for at most 20-30ms.
[18:56:34][V][muart_bridge:036]: Sending to heatpump [FC.42.01.30.01]09 83
[18:56:37][W][muart_bridge:046]: Timeout waiting for response to 42 packet.

KazWolfe, 2024-04-07:
Interesting. I don't necessarily think this is a *problem*, but curious that your system doesn't know how to handle [packet 09](https://github.com/Sammy1Am/mitsubishi-uart/wiki/(0x62)-Get-Response#type-0x09---get-run-status).

For now, I think it's safe to treat this as not a bug - but there may be some interesting implications from this as well.

Sammy1Am, 2024-04-07:
Yeah, no concerns that I would be aware of.  The only implication is that you'll have a 3-second delay occasionally while it's timing out waiting for that packet (so if you tried to change something during that window, you'd have to wait up to 3 seconds longer than usual for the system to respond).

Wouldn't be *too* hard to add a config in in the ESPHome config to selectively disable certain packets.

KazWolfe, 2024-04-08:
Wonder if it'd be good to scope in something like *n* sequential failures to get a packet removes it from the polling loop. Could be good to do around the same time we look at (properly) throttling certain request types.

Ricc68, 2024-04-08:
Thats an idea.
Another idea is: can this lack of 42 packet be used to autodetect something about the unit?
For example if the 42 packet is missing maybe we can infer that the unit supports a certain set of features and not others.